jon-harper/OmniBox #3: BIGTREETECH PITFT support?

Majawat, 2022-07-07:
Love this project!

Any way you could add support for the BIGTREETECH PITFT screens? Either the 4.3", 5", or 7" (but 5" or 7" preferred).

https://www.biqu.equipment/collections/lcd/products/bigtreetech-pi-tft43-v2-0-screen-board?variant=39840251936866.

Not sure if you'd keep the PI mounted to the screen, or keep it in the tray (probably tray? though that'd require a longer DSI cable, which should be fine.)

jon-harper, 2022-07-07:
Steve's original project has a set of 5" files that are easily adaptable and set the TFT on the lid above the display. I don't have access to a PITFT myself, but if you're willing to fit test, I'm willing to make the adjustments for you.

I'll also take a look at the 7" and see what I can do with that. Bear in mind the case is 7" wide, so it may look slightly awkward.

jon-harper, 2022-07-11:
I've got CAD files for the display and Steve's mounts as reference. It looks like both the 5" and 7" will mount fine. The longer DSI cable is going to be necessary, btw.

I should have something for you to test before end of week. Which display are you getting, or do you have one already?

jon-harper, 2022-07-11:
Do what's best for you, dude. If you want the 7", someone else will come along to test the 5". 🙂

Majawat, 2022-07-11:
Haha, I definitely get what you're saying, just a hard decision for me. Ok, I'm going to go with the 5".

jon-harper, 2022-07-19:
Heya, Majawat. Hope you're well. I'm almost done with some bugfixes and such to get 0.9.7 out the door. Getting your Pi TFT going is next on my design list.

The simplest approach is to attach a hollow pillar to the Raspberry Pi screw holes in the back of the TFT. The ribbon cable can run through a hole into the center. I'm planning to make this pillar part of a half-length case lid that will sit up front.

I want to ask a few questions before I get too far, though.

1. What mounting hardware does the TFT come with? The [Amazon page](https://www.amazon.com/BIGTREETECH-Direct-Capacitive-Octoprint-Raspberry/dp/B08F1WMQ1J) shows brass standoffs and mentions M2.5 screws but doesn't list what hardware comes it ships with.
2. Will the screen look okay without an outer protective panel?

Thanks!

Majawat, 2022-07-19:
Lovely! I got the 5" last week and attached it just to get KlipperScreen working to play with. 

1. No other hardware. Just the standoffs, screws, screen, and short cable. 
2. I think so! There's a small bezel around the screen otherwise no features. 
[image]
[image]
[image]
(Forgive the mess)

There is a small potential clearance issue with the display cable on the left. Or the brightness dial on the top right. 

Let me know if you need any other info

jon-harper, 2022-07-19:
Thanks a ton for the pics! Unexpected and super helpful!

Can you tell me how long those standoffs and screws are? I'm probably going to hijack them to help mount the screen, rather than using the outer screw holes.

Majawat, 2022-07-19:
[image]
[image]

jon-harper, 2022-07-23:
[PiTFT Test 1.zip](https://github.com/jon-harper/OmniBox/files/9173872/PiTFT.Test.1.zip)

This is a test for fit file. If it works, the rest of the mount should work and I'll send you a full STL for the lid/stand combo.

Steps:

- Add the standoffs to the TFT in the Raspberry Pi holes;
- The standoffs should fall in to the holes in the tray;
- You should be able to use the stock 2.5mm screws to then tighten it.

If anything doesn't go as planned, lemme know! I guessed at the size of the holes for the standoffs. Right now they're 6.5mm, which should be loose enough. If they could be tighter or don't fit at all, likewise let me know. 

Thanks again for testing this!

Majawat, 2022-07-23:
Seems pretty perfect. 

[image]

I looked at the standoff holes, they're slightly big, but it's stable without needing to be tighter.

jon-harper, 2022-07-23:
[Lid - PITFT.zip](https://github.com/jon-harper/OmniBox/files/9174036/Lid.-.PITFT.zip)

Awesome! Here's the TFT mount, then. This is a half lid so that you can customize the back half still. I haven't uploaded that template yet but will soon.

I hope I got the slot for the flat ribbon cable in the right orientation.

Majawat, 2022-07-23:
I think I'm a bit confused. Where is this supposed to go on the box? Up top like this?

[image]

I was thinking it would have gone in the front, like here instead:
[image]

Majawat, 2022-07-23:
[image]
Also, it would probably be better if the cable came in through the side instead of the top if possible.

jon-harper, 2022-07-24:
> I think I'm a bit confused. Where is this supposed to go on the box? Up top like this?

Yes indeed. This is the format for the Pi TFT that Steve's case uses, and I admit I took the placement from him. The normal printer display goes below so that they appear stacked at an angle with the Pi TFT above.

Quick render:

[image]

> I was thinking it would have gone in the front, like here instead: 

The 12864 or other LCD screen usually goes there. Are you planning to do entirely without? I can adapt the Pi TFT stand for the front easily enough, though I do still need to get the top version finished.

> Also, it would probably be better if the cable came in through the side instead of the top if possible.

Gotcha. I'll add a side cutout for the wire. Do you ever use the top cable port or any wiring on the other sides? Any issues with cable reach?

---

0.9.7 is almost packed up for release. When it's done today I'm going to create the 0.9.8 branch to add this and the MOSFET mount.

FYI, I also have a half-size lid to add that will go behind the Pi TFT screen and cover the back half of your case (with a handle, too).

Majawat, 2022-07-24:
> The 12864 or other LCD screen usually goes there. Are you planning to do entirely without?

Yep! I'm running Klipper on my RPI for an Ender 5 Plus. None of which supports the original screen without heavy software modifications (a different Klipper fork). So I'll only be using the PITFT. 

I see no issues with the top print, so I'd call that one good!

(Sorry for the close/reopening, my finger slipped as I typed)

jon-harper, 2022-07-24:
Awesome! I'll go ahead and commit the Pi TFT lid, but I'll leave this issue open. It shouldn't take too much time to attach the modify a display template now that the fit work is done. I'll plan to have that to you (sometime =) this week.

jon-harper, 2022-07-26:
Howdy Majawat. Hopefully this works and is what you were looking for.

closed 2022-08-18 by jon-harper