jon-harper/OmniBox #11: Mean Well RSP-500-24 support

Majawat, 2022-07-26:
The mounting holes for the Mean Well RSP-500-24 do not match the current Core files. It is larger than the current files, taking up majority of the lower base.

Bonus if there was a way to add an extra fan to the case to fool down this large PSU.

jon-harper, 2022-07-26:
> The mounting holes for the Mean Well RSP-500-24 do not match the current Core files. It is larger than the current files, taking up majority of the lower base.

Oof, my bad for not considering this. I even have an SE-450-24 on a printer and knew the mounting holes are not universal.

I'll get a CAD file and get to work on this.

> Bonus if there was a way to add an extra fan to the case to fool down this large PSU.

Couple of thoughts:

- The new generic rear panels have 60mm exhaust fan that might help.
- I will look at a 40mm mounted low on the rear panel (next to the power connector), as well.
- Alternatively, a big ol' 92mm mounted midway up the rear panel would solve all of your problems.
- As for intake, I can make the "teeth" along the bottom of the base taller. That should increase airflow, too.

jon-harper, 2022-07-27:
Couldn't find a CAD, but the spec sheet works as well. The fourth screw hole is a bit odd--the RSP-500 has an "extra" M4 screw hole in the middle. You'll be using that instead of the corner screw  hole near the Raspberry Pi (it would require a tray redesign). Here's what I mean:

[image]

As always, let me know if you run into any issues.

---

P.S. I haven't modified the base yet to increase the intake "teeth" size. I'm pondering the idea of a universal PSU mount, so I'm hesitant to do too much right now.

commit dbbd0920 (2022-07-27): Fixed #11 and added PSU type folders to Core

- Fixed #11
- Core now has two subfolders for PSU type

If anyone has ideas for a universal PSU mount, I'm all ears.
closed 2022-08-18 by jon-harper