jon-harper/OmniBox #19: Add tray support for Solid State Relay

kameron-m7, 2022-08-17:
Feature Request: Add support for Fotek SSR-40 DA relay

[https://www.amazon.com/SSR-40DA-Solid-Output-24-380V-SSR-40/dp/B07FVHWN82](Amazon)
[https://grabcad.com/library/ssr-40-da-1](GrabCAD)
[https://cdn.sparkfun.com/datasheets/Components/General/SSR40DA.pdf](TechSpecs).

Thanks!

jon-harper, 2022-08-17:
Let me know how it fits and any suggestions you may have. I'm unsure if the slots that support the sides of the SSR are adequate to keep it from rocking.

jon-harper, 2022-08-17:
I revisited the dimensions and realized I *can* fit it on a half tray! The latest commit should do it.

kameron-m7, 2022-08-17:
Awesome, Thanks!

jon-harper, 2022-08-25:
Heya @kameron-m7, how does it fit? I want to make sure all is well before I close the issue. 😃

kameron-m7, 2022-08-26:
Fits well for me. Thanks!

jon-harper, 2022-08-26:
Awesome!

closed 2022-08-26 by jon-harper