jon-harper/OmniBox #46: 2 Request TFT 50 panel and BTT UPS 24V modual

wizard311, 2022-10-24:
I looked through the files and saw TFT35 style panels but nothing as large as TFT50.  Would also like a module for BTT UPS 24 to put inside the OmniBox.  Also is there a way to add 120MM fan to the omnibox?

jon-harper, 2022-10-24:
> I looked through the files and saw TFT35 style panels but nothing as large as TFT50.

I'll grab the CAD file and take a look at how to mount it.

> Would also like a module for BTT UPS 24 to put inside the OmniBox.

Definitely. I took a look at the dimensions and it may or may not fit on a short lower bay tray--I might need to use a long one.

> Also is there a way to add 120MM fan to the omnibox?

Yes and no. I used them early on since I had quite a few spare, though they would only fit on the lid. Also, you can get great cooling with even a [single 80mm in the back](https://github.com/jon-harper/OmniBox/tree/main/Panels/Rear%20Panel/Custom/Ender%205%20Plus).

At the same time, you're not the first person to ask about 120mm fans, so I'll see what I can do.

wizard311, 2022-10-25:
Thank you for the fast response.  I am converting a Elegoo Neptune 3.  Ok thats a lie the only thing left of the neptune is the frame as I have changed it over to Exoslide y and xz for movement and BTT S40C X and Y closed loop steppers, Dual Z steppers, Octopus Pro and Raspberry Pi 4. Also changed over to E3D Revo and Orbiter Extruder. The Neptune 3 just doesn't have a lot of room to mount things so your OmniBox is a blessing at this stage in my conversions. Thanks again for this perfect answer I didn't know I needed.

jon-harper, 2022-10-25:
I'm glad to help! That sounds like an awesome machine you've got, and might become the first resin printer on an OmniBox.

I can't promise timelines on getting parts to you, but are you comfortable if I upload zipped STL files to this issue for you to print and fit test? I don't have the two BTT parts here, just CAD models.

Finally, a compatibility question: are you using a Mean Well LRS-350 power supply or something else?

wizard311, 2022-10-25:
I am fine with zipped STL files.  I have LRS-350 at the moment but I am looking to upgrade to RSP-500-24.

jon-harper, 2022-10-25:
Cool, just making sure. ✔

The next OmniBox version (0.9.9) coming out next month will have universal support for power supplies, by the way (you'll be able to remove the PSU and replace with a different model).

jon-harper, 2022-10-25:
[BTT TFT50 Test Fit.zip](https://github.com/jon-harper/OmniBox/files/9860946/BTT.TFT50.Test.Fit.zip)

I got really lucky here. The project that OmniBox is based on had a file for this that I could adapt. BTT didn't release a CAD model for this display, so I can't even do a cursory check for fit myself.

It might take some fussing to get this right, but it should be close to start.

jon-harper, 2022-10-25:
Your timing is pretty spot on. I closed out the 0.9.9-experimental branch and folded it into 0.9.9 a few hours ago. Any changes from here to release for the Core body are likely to be cosmetic.

Please be aware that 0.9.9 is still an active branch, too. I'm still making sure all of the new and changed panels and trays are exported and merged, as well as testing a few. Use at your own risk, as it were. 🙂

jon-harper, 2022-11-23:
Hey, I'm checking on all of the open issues for product support. I'm hoping to release by week's end and want to include everything tested to fit that I can. If you've fit any of the trays or panels requests, can you let me know how it went?

Many thanks!

wizard311, 2022-11-24:
I am still attempting to print all the parts for the OmniBox. I know the
display panel BTT50 doesn't fit in is current rendition.  I am not very
good with any type of CAD software so I inserted the panel in the front
part of the OmniBox and "deleted" the BTT50 from it to create the void
necessary to install the BTT50 in the real world.  I know this is backwards
from the way it's supposed to be done but again I am not talented in using
CAD software.

I haven't finished printing the other custom parts you have provided for
me.  My printer only prints at 30ms so these are taking days per print.
Once I have all of them printed I have another idea I want to run past you
but I have to get it assembled to test my idea.

Thanks again for all your hard work.

jon-harper, 2022-11-25:
Thanks for getting back to me. Just wanna make sure I have stuff in the right place before release. 🙂

I'll check on the BTT display. Sounds like I left the actual panel to attach it to the case. 🤦‍♂️