jon-harper/OmniBox #48: [MCU Tray] - BTT SKR3 & BTT SKR 3 EZ

francis-robert, 2022-10-26:
## Request details

Both the recently released SKR 3 and SKR 3 EZ boards have different screw positions than the other SKR 1.3+ boards.

- In the case of the SKR 3, there's only a 0.2mm difference, since they seem to have added a bit of padding.
- In the case of the SKR 3 EZ, however, it's trickier and would require a new layout since the board shifts the top-right screw hole to accommodate one of its EZ sockets.

The [BTT SKR 3 Repo](https://github.com/bigtreetech/SKR-3) contains the sizing PDF as well as step files to make easier modifications and design new parts for the boards.

### Affected Components

- MCU Tray

jon-harper, 2022-10-26:
Thanks a ton for catching this.

I'll get something put together shortly. Are you willing to fit test if I attach the results here?

francis-robert, 2022-10-26:
Yeah, I would absolutely do that. I only have the SKR 3 EZ and not the 'standard' SKR 3 board though. Then again, it's arguably the more important of the two to test.

I also added an edit to the top post since I just realized the documentation would also need to be updated to reflect that the current designs aren't actually all SKR 1.3+ boards and there may be subtle variations between SKR variants

jon-harper, 2022-10-26:
I really appreciate it. I will get a tray for the SKR 3 EZ to you today.

jon-harper, 2022-10-26:
[image]
[BTT SKR 3 EZ.zip](https://github.com/jon-harper/OmniBox/files/9871965/BTT.SKR.3.EZ.zip)

Here it is. You likely want the M4 version (`MCU Tray - BTT SKR 3 EZ - M4.stl`) unless you're an early bird and are printing an 0.9.9 case.

jon-harper, 2024-11-28:
I'm going to close this as complete unless I hear otherwise. It's been years.

closed 2024-11-28 by jon-harper