jon-harper/OmniBox #63: Request BIGTREETECH Relay V1.2 Bottom Plate

wizard311, 2022-11-13:
I am wanting to add a BTT Relay V1.2 to the Omnibox I used the file from this thingiverse as a blueprint and the bottom plate from "Lower Bay - DROK 5A Buck with LED" to start a file for it but I feel it could use supports for the screw posts and possibly wire ties like the "Lower Bay - DROK 5A Buck with LED".  I attempted to keep the mounting post at the same height to fall into your ecosystem.  Please feel free to make any and all changes to create this part for Omnibox ecosystem.  

https://www.thingiverse.com/thing:4428092
[BIGTREETECH Relay V1.2.zip](https://github.com/jon-harper/OmniBox/files/9997702/BIGTREETECH.Relay.V1.2.zip)

jon-harper, 2022-11-14:
Cool! I'll see what I can put together. This will give me a break from documentation writing, too. 🙂

jon-harper, 2022-11-16:
[Here's the folder in the unstable branch](https://github.com/jon-harper/OmniBox/tree/0.9.9-unstable/Trays/Lower%20Bay/BTT%20Relay%201.2).

I put a pair of zip tie anchors on one side, then decided to make it symmetrical. They anchors are really close to the edges so that you have room for that rather long board. If I moved them too far out and it's inconvenient, I can adjust.

Let me know how it goes. 👍

wizard311, 2023-04-08:
Test fit complete and perfect fit and function

jon-harper, 2023-04-08:
Thanks for letting me know!

closed 2023-04-08 by jon-harper