jon-harper/OmniBox #70: [MCU Tray] MKS Skipr

Killajoedotcom, 2022-11-25:
### UNTESTED. My submission for the MKS Skipr

Wanted to use a spare MKS Skipr motherboard but noticed only BTT stuff was supported. Generated this version for myself, but wanted to give back to the project because it's so good. I have yet to print and test this, but it uses M3 heatserts for longevity. 
[MCU Tray - MKS Skipr - Heatserts.zip](https://github.com/jon-harper/OmniBox/files/10091728/MCU.Tray.-.MKS.Skipr.-.Heatserts.zip)

[image]

If good enough, I hope this can be added to the project eventually.

jon-harper, 2022-11-25:
Thanks!

I made a note on the PR that I totally messed it up. I've cherry-picked the files into `-unstable` and will fix up a version with stock screw holes, too.

Let me know how the fit test goes; once I've done my end and the fit test is okay I'll close this.

jon-harper, 2022-11-27:
Your tray looks great! I believe your design is for the 4mm long inserts? They're much more common than the ones I use in the rest of the case so I kept your design and added one for the 5.7mm long inserts and one without inserts at all (like standard trays).

Killajoedotcom, 2022-11-27:
> 

Yes, I build Vorons quite frequently and have hundreds of the M3x5x4 heatserts on hand. Thanks for making other versions.

jon-harper, 2022-11-27:
> Yes, I build Vorons quite frequently and have hundreds of the M3x5x4 heatserts on hand. 

This...is something I very much should consider moving forward.

> Thanks for making other versions.

Thanks for suggesting the board *and* making the first one!

jon-harper, 2023-01-21:
Got this in the docs now. Closing.

closed 2023-01-21 by jon-harper