jon-harper/OmniBox #82: BOM SD extender not fitting?

phillyaws, 2022-12-08:
Hi!  Printed out the [SD Card extension](https://github.com/jon-harper/OmniBox/tree/main/Misc/Lanmu%20MicroSD%20Extension) for the front panel, and the [extender in the BOM ](https://smile.amazon.com/LANMU-Extension-MicroSDHC-Monoprice-Raspberry/dp/B07WWVBK8V/ref=sr_1_3?keywords=micro+sd+card+extender&qid=1670512617&sprefix=micro+sd+card+ext%2Caps%2C68&sr=8-3)doesn't seem to fit.  Is the BOM wrong, or am I?

jon-harper, 2022-12-08:
Are you using an SKR 1.4 or 2 by chance? Someone testing it found that problem about a week ago(issue #79 )

If it's another board...new bug!

In the mean time, [this has a longer extension](https://www.amazon.com/gp/product/B07YYSP5F5) as a workaround, with my apologies.

phillyaws, 2022-12-08:
Oh, I mean the physical card...holster?  Receptacle?  I'm not sure of the technical name, but the black plastic part of the extender is too big for the file mount.

jon-harper, 2022-12-08:
Ah. No worries. Pry it open and remove the PCB inside. :-)

jon-harper, 2022-12-10:
Hey, I'm just making sure that this fixed the problem for you before I close the issue.

I'm going to either figure a way to illustrate that step or make it a visible note. It's easy to miss since it's just a line of text without a graphic.

phillyaws, 2022-12-10:
> Hey, I'm just making sure that this fixed the problem for you before I close the issue.
> 
> I'm going to either figure a way to illustrate that step or make it a visible note. It's easy to miss since it's just a line of text without a graphic.

I'm going to check when I get home today!

phillyaws, 2022-12-12:
It worked perfectly!  Now I just need to figure out how to use Fusion to make rear/sideplates, and I'm home and hosed!

closed 2022-12-12 by phillyaws

jon-harper, 2022-12-12:
Awesome! Thanks for checking back, too!

It may not be what you need, but opening one of the existing design files
will demonstrate a working example that you can modify. Good luck!