jon-harper/OmniBox #87: MCU Tray - Fysetc Spider

MaffooClock, 2023-01-08:
I've used the template to design a MCU tray for the [Fysetc Spider](https://wiki.fysetc.com/Spider/).

[image]

I've opened PR #86 for you to review and consider.

jon-harper, 2023-01-08:
Thanks! Have you had a chance to fit test yet?

MaffooClock, 2023-01-08:
I have not, but I am printing it at this very moment, and will share a photo of the MCU mounted and the tray inside the body.

jon-harper, 2023-01-08:
Rock on, thanks!

MaffooClock, 2023-01-08:
Wahoo, it works!  Even though I haven't finished assembling my OmniBox, I don't see any clearance or wiring issues at this point.

[image]
[image]

jon-harper, 2023-01-08:
Sweet! I'll get this merged and in the docs; leaving it open for now (so I don't forget). Thanks for the design!

jon-harper, 2023-01-21:
It's in the docs. Closing.

closed 2023-01-21 by jon-harper