jon-harper/OmniBox #88: BTT PI TFT43/TFT50/TFT70

MaffooClock, 2023-01-10:
[The documentation](https://jon-harper.github.io/OmniBox/0.9.9/support/displays/#raspberry-pi-displays) lists the BTT TFT43/TFT50/TFT70 product family specifically, and I purchased a TFT50 so that I could have it in the front panel rather than on the lid.

I see you've added a display panel for the [PI TFT50](https://biqu.equipment/products/bigtreetech-pi-tft43-v2-0-screen-board?variant=39337700819042) in commit a3d2cb5, but I'm confused by [the design](https://github.com/jon-harper/OmniBox/blob/f1a972f000c8eb2b4fda452bd85436bae8aea70d/Panels/Display/BTT%205.0%20TFT/Display%20Mount.stl).  It appears to be similar to the smaller LCDs that include a knob, however, the PI TFT43/TFT50/TFT70 family are just capacitive touch displays (no knob).

I must be confused, and the [Panels/Display/BTT*](/jon-harper/OmniBox/tree/unstable/Panels/Display) files must be for a different product family.  Are the TFT43/TFT50/TFT70 files somewhere else and I'm just not smart enough to find them?

For reference, this was discussed in these issues:
  - #3 
  - #46 

Not complaining!  Figured I'd stumbled into some ambiguity between panel files and documentation that ought to go on the to-do list 🙃 

As I mentioned, I have a [TFT50](https://a.co/fCm2DUI) coming (should be delivered today or tomorrow), and I'd be happy to help with fit-test or even designing a panel for it to contribute here.

By the way, thanks for sharing such a fantastic project, and for being so helpful and involved.  I am eager to contribute as much as I can.

jon-harper, 2023-01-10:
TL;DR: The [Raspberry Pi Display](https://github.com/jon-harper/OmniBox/tree/main/Panels/Display/Raspberry%20Pi%20TFT) is what you're looking for. It mounts using the Raspberry Pi screw holes on the back of the display, so any screen  meant for an RPi works with it.

---

Ah! There are two different products called TFT50 and you've *definitely* stumbled on some documentation ambiguity.

There's the ["BTT TFT50"](https://www.amazon.com/BIGTREETECH-Direct-Printer-Display-Screen/dp/B08B34CXTV), which is for MCU displays and is a touchscreen plus knob. Then there's the ["BTT PI TFT50"](https://www.amazon.com/BIGTREETECH-Screen-Display-800x480-Raspberry/dp/B08FD2YZ23), which is is *just* a touchscreen as is for the Raspberry Pi.

The confusion about which product it references is that:
- the plain TFT50 display isn't documented yet (an oversight)
- the docs don't link to the PI TFT50 files (ditto).
- The GitHub readme is also way out of date and mentions none of these (ditto refrain).

I've tagged this issue as a doc bug and am leaving it open until I've reworked that section of the documentation.

-----

> By the way, thanks for sharing such a fantastic project, and for being so helpful and involved. I am eager to contribute as much as I can.

Thanks! I really appreciate that. And thank you for contributing, including finding this issue; every hand helps.

MaffooClock, 2023-01-10:
> The [Raspberry Pi Display](https://github.com/jon-harper/OmniBox/tree/main/Panels/Display/Raspberry%20Pi%20TFT) is what you're looking for...

💡 Ah.  I saw that, but kept thinking it was specific to the [official Raspberry Pi Touch Display](https://www.raspberrypi.com/products/raspberry-pi-touch-display/) (which I understand would be a top-mount only; I was confusing myself).  Further, I was a bit thrown off by how it's just a flat panel with vent holes, so I never gave it any further thought.

> ... It mounts using the Raspberry Pi screw holes on the back of the display, so any screen meant for an RPi works with it.

👀 Now I see that the standoffs on the display match up with mounting holes in the panel.  It's a bit odd that the display will sit naked on top rather than inside a bezel, but I suppose the raw look isn't so bad.

> There's the BTT TFT50... Then there's the BTT PI TFT50

🎉 Aha!  Okay, that clears it up -- I did a smidge of Googling before I opened this issue, trying to see if there was something different, but I only saw the Pi-only version in results (I didn't try very hard, tbh).

Groovy, thanks!

I may try to make some time to design a panel that is specific to the BTT PI TFT43/TFT50 that lets the display sit flush behind the panel, if that would be welcome.  The [Raspberry Pi TFT](/jon-harper/OmniBox/tree/main/Panels/Display/Raspberry%20Pi%20TFT) panel is fine, but since this project is about options...

jon-harper, 2023-01-12:
> It's a bit odd that the display will sit naked on top rather than inside a bezel, but I suppose the raw look isn't so bad.

Apologies for that; I don't own an RPI screen, so I took the lid mount and spliced it on a display panel--the only thing I could be sure wouldn't have clearance issues.

> Groovy, thanks!

Glad to help! Thanks for pointing out the doc issues. I knew that area needed work, but hadn't looked at it in awhile. Out of sight, out of mind...

> I may try to make some time to design a panel that is specific to the BTT PI TFT43/TFT50 that lets the display sit flush behind the panel, if that would be welcome. The [Raspberry Pi TFT](https://github.com/jon-harper/OmniBox/tree/main/Panels/Display/Raspberry%20Pi%20TFT) panel is fine, but since this project is about options...

🤝🤝🤝 Flush mount designs are definitely welcome! I love a universal mount, but I love a nice fit better (and plenty of options, too).

MaffooClock, 2023-01-13:
I've got a rough design, but I think it's pretty dang close.  The only concern I have is the depth, namely whether the rear shell will intrude on the lower bay trays area.

I'll print and test the fit, make whatever tweaks may be necessary, then hopefully be close to opening a PR...

[image]
[image]

jon-harper, 2023-01-13:
Flush mount? Nice!

MaffooClock, 2023-01-14:
Success!  First print, everything went together smoothly, and I had no clearance issues behind it.  The mounting ears on the rear shell require supports, so I made them kinda beefy and kept them flat to make removing supports easy and reduce risk of breaking.

BOM:

1. 8 x M2.5x6 BHCS or FHCS for attaching display to rear shell (could get away with just 4)
2. 4 x M3x5ODx4L heat-set inserts (for HSI version)
3. 4 x M3x6 BHCS or FHCS for attaching rear shell assembly to front panel

This is just for the Pi TFT50; no plans at the moment to do the Pi TFT43, maybe we'll see if anyone asks for it.

If you dig it, I'll open a PR for you to review.

[image]
[image]
[image]
[image]
[image]

jon-harper, 2023-01-14:
I am all about this. Whenever you feel it's ready, open that pull request!

Side note: formal support for M3x5x4mm inserts is coming.

jon-harper, 2023-01-27:
Reopened this so I don't forget to add to the docs.

MaffooClock, 2023-01-27:
I thought I'd take a crack at the documentation, too.  See commit [6b9dca5](https://github.com/MaffooClock/OmniBox/commit/6b9dca5ca3b3cddd1df16bf05d8939f4c6262524) to review.

Aside from just adding the relevant section, I also added a note to the "Raspberry Pi Displays" section.  Not sure how you'd feel about that -- maybe you don't want the note, or maybe you might remove this display from that section altogether.

If you're okay with going this route, then add comments to that commit of any changes you'd like to see, then I'll open a PR afterward.  Or I could open a PR as-is, if it's mostly right, and I guess you can make edits before accepting the PR (I'm not a GitHub expert)?  Or heck -- copy/paste what I've done as a starting point if that'd be easier on you (I don't care about credit, just want to help).  Or purple, because ice cream has no bones!

jon-harper, 2023-01-27:
I think I can cherry-pick across forks, so I will try that. Regardless of how I add it, your work is definitely helpful and very much appreciated.

jon-harper, 2023-02-16:
Okay, it's documented. I completely reworked that section of the docs, too. It should be a lot clearer now.

closed 2023-02-16 by jon-harper