jon-harper/OmniBox #91: Display template is faulty

jon-harper, 2023-01-10:
A user reported that the display template is has the wrong screw hole locations. The template is inherited from Steve's project, so it very well may never have been updated.

MaffooClock, 2023-01-17:
I used this template for my BTT Pi TFT50 display (#88), and while those components fit together perfectly, I did notice the mounting holes in the corners were slightly off from those in the main body; ~1-2mm vertically inward (horizontal spacing lines up fine).

I started to bring this up, but figured it might be possible that my main body component warped slightly during printing, so I didn't want to say anything until I was certain.  I wanted to check the design file for the main body to see what the vertical distance is between mounting holes to compare with those in the template (and my designed based off that), but I don't see a .f3d or .step file.

I'll hold off on submitting a PR #88 until we can confirm correct spacing.

jon-harper, 2023-01-17:
Thank you for checking on this; there are a couple of entangled display bugs/requests here I need to sort out. I intend to pare down the bug list later this week, hopefully knock this one out, and get some new stuff merged. :smile:

closed 2023-01-22 by jon-harper

MaffooClock, 2023-01-23:
Now that you got this sorted out, what is the vertical distance between screw holes supposed to be?  As a test, I re-printed my BTT Pi TFT50 with 103mm vertical distance, and it was spot-on (horizontal distance was unchanged).

I'll submit a PR for the new display panel once I know the correct measurement 👍

jon-harper, 2023-01-23:
103.825mm 😊

commit b267f64a (2023-01-22): Fixed #91

Screw holes were definitely off, probably from v0.9.5 -> 0.9.7.
commit 057ea4d5 (2023-01-22): Fixed #91

Screw holes were definitely off, probably from v0.9.5 -> 0.9.7.