jon-harper/OmniBox #94: Add a power switch cover to prevent accidental bumping

jon-harper, 2023-01-17:
The power switch is too easily bumped and can power down the printer on accident.

Solutions:

1. Add a switch cover plate
2. Implement the suggestion of a hard-throw switch that won't flip so easily.

MaffooClock, 2023-01-17:
Funny... I was just thinking about this the other day.

My thought would be to add guard rails on either side.  It could be an add-on piece, so the base would have an alternate design that would allow this extra piece to be affixed over the switch.

Something similar to this (doesn't need to look like this, I just found it in a Google Images search as an example):
[image]

This design makes it easy to turn on, difficult to turn off (accidentally).

jon-harper, 2023-02-09:
Hey Matthew. [This commit}(https://github.com/jon-harper/OmniBox/commit/72bce84f87fa234fad469b4650a23c19bb79bdec) seems to do the job if you want to give it a shot.

The power switch needs about 1mm clipped from the ends of the snap-in "ears" to make room for the cover.

MaffooClock, 2023-02-09:
Well done 👍  I will definitely print one and check it out!

jon-harper, 2023-02-16:
Switch cover is now [merged in main](https://github.com/jon-harper/OmniBox/commit/48f5374f79bf7ff902ebda646f5c765db5af0da4). I'll close this after the docs are done.

jon-harper, 2023-05-17:
I've added the switch to all of the 0.9.10 docs but the assembly portion.

closed 2023-05-17 by jon-harper