jon-harper/OmniBox #96: Would love to see a tray for a 5V PSU

piotrb, 2023-01-31:
I have a Mean Well RS-25-5 PSU kicking around this thing is quite small, it could easily fit on a tray .. 

This could be a nice semi standard config for running the CPU on a separate power supply (so you can turn off the big power supply when not needed)

Here is the datasheet with dimensions: https://www.meanwellusa.com/upload/pdf/RS-25/RS-25-spec.pdf

Thoughts?

jon-harper, 2023-01-31:
There is a user who was working on this last week and discussed it in Discord. I've sent them a ping. I do think this is a nice add, but one I haven't been able to do without buying one.

jon-harper, 2023-01-31:
Here's what they are working on (user edgy on the OmniBox Discord):

> oh, yeah.  i haven't printed my design yet but based on the datasheet and dummy model in freecad it should work.  the rs-25-5 is wider than one of the short trays so i extended it a little with the intention of placing it in the center of the lower bay using the extra space towards the back (probably fit in the rear bay also, but i'm planning to place a long tray with wagos there).  the fan mount was made super simple (and probably not thick enough for the screws) so it could be skipped if not needed.
> the rs-25-5 could easily fit on a long tray and i haven't sketched that out yet since it didn't fit my initial use case but wouldn't take long to do.

[image]

In short, it's in progress. :-)

jon-harper, 2023-02-27:
edgy published his mod [here on Printables](https://www.printables.com/model/388529-omnibox-lower-bay-trays-for-mean-well-rs-25-5).

I'll close this once I've got the docs caught up.

closed 2023-03-23 by jon-harper