jon-harper/OmniBox #105: [REQUEST] BIGTREETECH TFT35 SPI V2.1 Touch Screen Display Mount

YFNOtaku, 2023-03-22:
https://biqu.equipment/collections/lcd/products/bigtreetech-tft35-spi-v2-1-touchscreen-io2can-module

I have one of these in the mail now to add to my system, and was hoping someone with more modeling skill than I (which is none currently) could help me with this. Don't really care if it's a front or top mount, I would appreciate either. Also found a case that you could use as a potential starting point. Thanks in advance!

https://www.printables.com/model/403192

jon-harper, 2023-10-13:
I was going through old issues and realized I never replied to this. I was holding off replying until I could find out if anyone has one of these to model against, and then forgot. I am so sorry!

Display mounts are a little more involved to design than, say, an MCU tray, where you have just four holes and no real clearance issues. I don't have this display to work against, but I can try via CAD alone if you're still in need of a display mount and can test.

jon-harper, 2024-11-28:
I'm going to close this as wontfix. I'm open to contributions, however.

closed 2024-11-28 by jon-harper