jon-harper/OmniBox #108: Fit Test: New RSP-500 PSU tray

jon-harper, 2023-05-09:
This version is modified to make room for M4 socket head cap screws.

[Files are in the `unstable` branch.](https://github.com/jon-harper/OmniBox/tree/unstable/Trays/PSU/Mean%20Well%20RSP-500)

cereal7802, 2023-05-13:
I don't have M4 socket caps to test with right now (they are coming), but the tray works fine. The only issue I had was that with fans(4010) mounted in the PSU tray area, the psu does not fit as it is longer than the LRS-350. It does fit fine with no fans mounted and probably doesn't need them as the rsp-500 has a fan built in it that moves a decent amount of air.

jon-harper, 2023-05-13:
Thanks for checking! I'm going to hold this open until I can verify the socket head screws fit, just to be sure.

Re: the tray fit

The RSP-500 pretty much maxes out the space below. Since the fan up front blows out, I moved it forward to let the existing fan try and do the work (and try and leave more room at the back).

There's a work in progress to center the PSU fan exhaust on the fan holes, but that's still in planning.

jon-harper, 2023-07-20:
I'm going to close this for now. CAD shows there's plenty of room for socket head screws and cereal's already made sure it fits.

closed 2023-07-20 by jon-harper